gauteh/omb-nano #13: Next PCB

gauteh, 2024-07-29:
* Correct space between pins for RockBlock
* More parallel battery-terminals

zerblatt007, 2024-09-10:
>    - Use ambiq module
>    - transistor for toggling GPS power ***@***.***
>    - usb to serial for ambiq module

Do we have any preference on usb to serial chip (for ambiq module)?

zerblatt007, 2024-09-10:
> Here is the one on the artemis nano redboard:
> image.png (view on web)

Makes sense to use that one, yes..

Roger

zerblatt007, 2024-10-25:
Have most of this in schematics form now..  
We want the fuse and perhaps a regulator?
[image]

zerblatt007, 2024-10-25:
Do we need something like this?
[image]

zerblatt007, 2024-10-25:
My reference is this, but maybe not a good one?
https://cdn.sparkfun.com/assets/5/5/1/6/3/RedBoard-Artemis-Nano.pdf

gauteh, 2024-10-25:
Can we use the pololu regulator that we already have on? We know it works with the modem, and it takes a range from about 2.5V to 11.5V?

zerblatt007, 2024-10-25:
> Can we use the pololu regulator that we already have on? We know it works with the modem, and it takes a range from about 2.5V to 11.5V?

Yes, of course.. It is set for 3.3V?

gauteh, 2024-10-25:
Yes. It's not configurable. Always 3.3V.

zerblatt007, 2024-12-03:
This is the amplifier used on before the GPS antenna: (BGA725L6E6327FTSA1) https://www.infineon.com/dgdl/BGA725L6_V2+0.pdf?folderId=db3a30431f848401011fcbf2ab4c04c4&fileId=db3a30433784a0400137ef9a4d341f23

It is controlled with the VACT_GPS_OUT on sfy and now maybe on the OMB-nano (U21)

[image]

Could it be that this pin is not controlled correctly on sfy and the reason for no GPS success?
Will there be a problem controlling it on the OMB-nano?

zerblatt007, 2024-12-03:
"Correct space between pins for RockBlock"
I have problems finding the correct footprint for RockBlock 9602 (9603 seems to be the replacement and has a different connector?)
So to use the 9602 we need to make a new footprint or line up 2 headers. But what is the correct distance between pin 6 and 8 (on the old schematics. I have a new symbol loaded, but the footprint is like the old one, just the symbol is changed..

gauteh, 2024-12-03:
Here are the docs for the 9602: https://docs.groundcontrol.com/iot/rockblock/specification/dimensions, the 9603 has a trickier connector. There are some CAD models there, but I don't know if they include the pins.

I _thought_ that the antenna was not active, but I'm not sure. It seems the SFY/notecard is supposed to always supply VACT_GPS_OUT, but maybe we need to connect VACT_GPS_IN to it: https://dev.blues.io/datasheets/notecard-datasheet/note-nbgln/#active-gps ? I don't see any hardware way to control this, but maybe there's an inductor that allows both active and passive to be connected? Or the middle pin in the U.FL. connector may not be used on passive GPS antennas.

gauteh, 2024-12-03:
> The AGT uses the 9603, and all the schematics and Eagle files should be here: https://www.sparkfun.com/products/18712 under the "Documents" tab :) .

I was thinking that we should try to get it to work on the 9602 first, since they have the same driver and same operating modes. But the 9602 has a much easier pin interface to develop with. I'm not completely sure what makes 9603 better?

We are building 10 OMB's for UiB in the early spring, maybe we can use this version then.

gauteh, 2024-12-03:
> I have problems finding the correct footprint for RockBlock 9602 (9603 seems to be the replacement and has a different connector?)

Actually, if this saves time to do right away we could go straight for the 9603?

gauteh, 2024-12-03:
> Depends on how many 9602 modules we have in stock? They are expensive?

We only have one now. I can order a 9603 if we go for that one. Plus we may have a few 9603's from broken OMB's that we can salvage.

zerblatt007, 2024-12-03:
Cable is probably a good idea, so things won't rattle apart, so then we need to find a suitable 10p socket for the PCB and a cable length of like 4cm (female-female)?
This is Molex 51021 10p, but this is not for PCB mount?:
https://no.mouser.com/ProductDetail/Molex/51021-1000?qs=zXrbR4Jv0OdnHhceUiV3oQ%3D%3D

gauteh, 2025-02-27:
[image]

closed 2025-02-28 by gauteh