gauteh/omb-nano #15: Next PCB (from v1.4)

gauteh, 2025-02-28:
- [ ] Swap polarity on Iridium CN1 connector

gauteh, 2025-02-28:
Possibly swapped transistor:

[image]

gauteh, 2025-03-03:
Tried to solder above the GPS and to VCC on r14 and transistor. This resulted in the serial to reset and restart every time I opened a serial connection. R14 is connected to rts on serial and reset on ambiq, so it's likely something went wrong there... Couldn't find any issues with the multimeter though. Might be worth starting from scratch.

gauteh, 2025-03-03:
Maybe it is resetting itself when turning on the gps? 😊

gauteh, 2025-03-04:
When I don't toggle the GPS_WAKE pin (D38 on ambiq) the ambiq doesn't reset anymore... don't know why.

gauteh, 2025-03-04:
It used to be, but may have cut it in trying to mod the PCB.

gauteh, 2025-03-05:
[image]

I think you got it right @zerblatt007! did you flip r6 as well?

gauteh, 2025-03-05:
Current use @ 3.3V:

GPS fixing: 25 mA
Iridium charging: 250 mA (can go up to 500 mA)
Iridium operation: 70 - 250 mA
IMU collection: 12-13 mA

power-down: 13 mA (possibly due to usb-to-serial?)

gauteh, 2025-03-05:
Maybe it's possible to put it in usb-suspend:

[image]

gauteh, 2025-03-06:
I added a power indicator:

[image]

@zerblatt007 : could you check if it is correct?

gauteh, 2025-03-06:
So the regulator on the CH340 is now mainly to ensure that it is off when no USB is connected? A bit risky to add another regulator now?

zerblatt007, 2025-03-06:
> So the regulator on the CH340 is now mainly to ensure that it is off when no USB is connected? A bit risky to add another regulator now?

Hehe, it is not a very complicated circuit, but any change could lead to something unknown, that is true.

We could perhaps have done this without any capacitors as this is only to be used in the lab, but maybe we would like more there later..

[image]

zerblatt007, 2025-03-06:
Maybe this would do?

[image]

gauteh, 2025-03-07:
ordered!

closed 2025-03-07 by gauteh