gauteh/omb-nano #19: Next PCB (from v1.5)

gauteh, 2025-03-06:
Ref #15 

- [ ] Consider removing jumper for VBAT / VUSB.

zerblatt007, 2025-03-06:
We could do the diode trick from SFY3, there is a lot of space so should be easy..

zerblatt007, 2025-03-06:
But it will use some power..

[image]

gauteh, 2025-03-25:
[image]

Now running window of my office (also with GPS signal..). 4.83V, single 1x3 D-cells (Japcell).

zerblatt007, 2025-03-27:
Should we check if those battery connectors have a PCB mount option?
If there are, we could get them pre mounted?
What is the name of the black connectors we solder on?

gauteh, 2025-03-27:
> What is the name of the black connectors we solder on?

https://www.sparkfun.com/led-strip-pigtail-connector-2-pin.html

closed 2025-05-20 by gauteh