gauteh/omb-nano #22: ambiq doesn't start when on battery power

gauteh, 2025-03-25:
* measure 1.25 on ch340 out when no usb connected
* attaching USB cable and connecting with tty starts the ambiq
* cutting power to ch340 makes battery power work.

gauteh, 2025-03-25:
* test if adding 10k resistor from v_ch340 to GND helps.

zerblatt007, 2025-04-01:
10K resistor did not help, but replacing AMS1117 with HT7333-1 did.
It has been replaced on v1.6 together with bigger battery terminals.

gauteh, 2025-04-01:
Great! Looks like we are approaching yet another version :) Do you want to make some changes to the antenna as well before doing a new round?

zerblatt007, 2025-04-02:
Yes, I was thinking I had a conversation with opinions with the antenna supplier, but maybe not..
So I should just move the edges a bit as they propose and we try again, I suppose..

zerblatt007, 2025-04-02:
I have adjusted this a bit and poured GND on the top layer as Ignion suggested.
Hope that helps, so what we miss is more holes for strips?

gauteh, 2025-04-03:
Yes, see https://github.com/gauteh/omb-nano/issues/19.

closed 2025-04-03 by gauteh